jaapp/smart-grind-by-weight #6: Mavin 300g LoadCell measuring wrong way

Joshude, 2025-09-18:
Hello and thanks for this awesome project!

I just wired everything together to make first testruns, if everything is wired corectly.
First "problem" was, that my Specialita seems to be wired the right way, so black to - and red to +. Unfortunatly i didn't test this beforehand so just put the connector to the waveshare and powered it on. After nothing happened i quickly turned everything off and switched + and - and now it runs. 

I have put the loadcell as you've shown ("Mavin" top, arrow pointing down, this is the only working position with the 3d printed files anyway. But when i tested it there was an error on the display. So i just tested to put the pressure from below (pointing up) and it showed the right values. 

The Loadcell is wired correctly 
```
E+    →    Red
E-     →    Black
A-    →    White
A+    →    Green
```

Maybe this is the reason for the second issue: The motor won't start. But i have the Eureka disassembled, i don't know if there is a safety-feature that prevents spinning the motor when the case is open? I couldn't see any switches for that. 
Is it possible that grey and white are switched, too? So that i need to switch them? 

Thanks

Joshude, 2025-09-18:
Okay... to add: Maybe i just should read and fully understand the instructions. After calibrating the LC the weighting now works. 

Only issue that stays is the motor not running

jaapp, 2025-09-18:
My bet would be that you need to connect the other wire. I think it is safe to conclude now that the wire coloring should be ignored. You can safely switch the button / motor wire around as this won't damage anything. 

Fortunately the Waveshoard board seems to have reverse polarity protection as i also discovered in practise...

Please report back. I'll change the doc to wire position instead of coloring.

jaapp, 2025-09-21:
OK well, i'll consider this issue closed then.