jaapp/smart-grind-by-weight #13: Stuck in predictive

dthomasen, 2025-09-28:
Thank you so much for this project!

I have successfully installed the mod on my Mignon Perfetto.
I used another load cell as I had one lying around and adopted the 3D print to that instead.

I have an issue where the grinder will start for about 7 seconds, make it to 14 gram (out of 18.5) and then stop. The display then shows the error "Stuck in predictive".

What could be the issue? Or anything I could try?

dwffls, 2025-09-28:
The latest commit had a small issue that the uses a mock loadcell driver. 

Could you try changing this line, found in hardware_config.h:
'''
#define HW_ENABLE_LOADCELL_MOCK 1
'''

To: 
'''
#define HW_ENABLE_LOADCELL_MOCK 0
'''

And build and upload again to see if that fixes your problems?

dthomasen, 2025-09-28:
Maybe worth mentioning that the load cell are a 1 Kg model, as I plan to modify it for portafilter use later.. 😄

dwffls, 2025-09-28:
What do you mean with "going up and down about 100"?

When calibrating the final screen should say 0g with nothing on the platform. Any other weight you put on it should show that weight on the display. 

If this is not the case I believe something is wrong with your loadcell/adapted housing. The fact that is made for 1 kg shouldn't be any issue.

I've had a similar issue with incorrectly displayed weights after adjusting the 3d prints. My dosing cup didn't fit so I adjusted the 3D prints files. However these new parts had some interference with the back cover, this wildly varied the output of the loadcell.

dthomasen, 2025-09-28:
The weight on the final screen of the calibration is 0 and is not fluctuating. But the first screen on the calibration fluctuates up and Down by about 100 and isn't settled on a single number.

I had the same issue as you with previous prints, but this one fixed it besides the above mentioned fluctuation

dthomasen, 2025-09-28:
Happened at 18.2 this time 🙂
[image]

jaapp, 2025-09-28:
Sorry for the slip up. I messed something up in my last commit causing development settings to be active. It's fixed now.

>  But the first screen on the calibration fluctuates up and Down by about 100 and isn't settled on a single number.

These are raw values indeed. They fluctuate a lot which is normal signal noise. Perhaps it is better to hide these RAW readings, i fully understand the confusion this causes. The intention was to give _some_ kind of feedback

> Happened at 18.2 this time 🙂

The grind timeout is a safety feature so we don't burn out the motor / empty the hopper if for whatever reason it's grinding unattended and something bad happens.

It is set to 30s. So either your grinder has a real low flow rate and 30s, or the taring process is taking ages somehow.
Can you share a video or a screenshot of a grind log session report (with taring process enabled in the visual)?

dthomasen, 2025-09-28:
> These are raw values indeed. They fluctuate a lot which is normal signal noise. Perhaps it is better to hide these RAW readings, i fully understand the confusion this causes. The intention was to give _some_ kind of feedback

Really appreciated that feedback during hardware debug though! Perhaps it's better to move it to some kind of status/debug menu

> The grind timeout is a safety feature so we don't burn out the motor / empty the hopper if for whatever reason it's grinding unattended and something bad happens.
> 
> It is set to 30s. So either your grinder has a real low flow rate, or the taring process is taking ages somehow. Can you share a video or a screenshot of a grind log session report (with taring process enabled in the visual)?
> 
> Or and please check to make sure the cup is not rubbing against the frame somewhere.

Ahh makes perfect sense! The taring process is indeed taking ages (probably around 10-15 seconds). There's no rubbing of the cup.

Where can I find this grind timeout in the code? I don't really need it for my use case, so I think I'll just increase it enough. 😊

I'll try to get a video and a grind log session when I get home. Thank you so much for the quick answers!

jaapp, 2025-09-28:
You sensor might be particularly noisy somehow. Did you hook up the ground wire of the loadcell as well? I see i never mentioned this in my wiring diagram so that's something I have to add on my end.

See if this resolves the issue. Shortening the length of the wire will help as well. 

Otherwise you'll have to increase USER_SCALE_SETTLING_TOLERANCE_G - this determines how high the weight may fluctuate while still counting as a stable reading.

Please report back, if more users are having this issue we might have to adjust the default value so it's more forgiving dor noisy setups.

jaapp, 2025-09-28:
update: USER_SCALE_SETTLING_TOLERANCE_G has now been renamed to GRIND_SCALE_SETTLING_TOLERANCE_G

dthomasen, 2025-09-29:
Shortening the wires for the load cell worked! Taring seems much faster now and grinding is within the default 30 second timeout.

Thank you so much for all the help! Feel free to close this :-)

closed 2025-09-29 by jaapp